balena-os/balena-jetson #10: [origin/wip_warrior_tx2]  Build Failed in tegra186-flash-dry_32.1.0.bb:do_configure

gancev, 2019-06-24:
NOTE: recipe tegra186-flash-dry-32.1.0-r0: task do_configure: Failed
ERROR: Task (//balena-jetson-nano/tx2build/../layers/meta-balena-jetson/recipes-bsp/tegra-binaries/tegra186-flash-dry_32.1.0.bb:do_configure) failed with exit code '1'

the problem in do_configure:

>     bspatch newboot.img boot0.img ${WORKDIR}/boot0.bindiff
>     cp boot0.img ${DEPLOY_DIR_IMAGE}/bootfiles/boot0.img

even when manually running the command i get corrupt patch

/balena-jetson-nano$ bspatch ./tx2build/tmp/work/aarch64-poky-linux/tegra186-flash-dry/32.1.0-r0/tegraflash/newboot.img boot0.img ./layers/meta-balena-jetson/recipes-bsp/tegra-binaries/files/boo
t0.bindiff
bspatch: Corrupt patch

floion, 2019-06-25:
Hi. What tag are you building?

gancev, 2019-06-25:
on branch 
**origin/wip_warrior_tx2**
same error on
**origin/tx2_update_and_nano**

bspatch is from 

> /home/user/balena-jetson-nano/tx2build/tmp/work/aarch64-poky-linux/tegra186-flash-dry/32.1.0-r0/recipe-sysroot-native/usr/bin/bspatch

the same error when manual execution
> ./bspatch /home/user/balena-jetson-nano/tx2build/tmp/work/aarch64-poky-linux/tegra186-flash-dry/32.1.0-r0/tegraflash/newboot.img boot0.img /home/user/balena-jetson-nano/layers/meta-balena-jetson/recipes-bsp/tegra-binaries/files/boot0.bindiff
> Failed to apply delta (-1)

floion, 2019-06-25:
@acostach is wip_warrior_tx2 still good for a PR? Or can it be deleted?

klutchell, 2019-06-28:
I'm experiencing the same issue as @gancev on the [origin/wip_warrior_tx2] branch.

The `bspatch newboot.img boot0.img ${WORKDIR}/boot0.bindiff` step of `tegra186-flash-dry_32.1.0.bb:do_configure` is failing.

```
| 2122752+0 records in
| 2122752+0 records out
| 2122752 bytes (2.1 MB, 2.0 MiB) copied, 13.5125 s, 157 kB/s
| Failed to apply delta (-1)
```
I'm using https://github.com/balena-os/balena-jetson-nano/commit/3ad689f7e9e2dbb9ee68978a8e89780d0a944e54

[log.do_configure.23539.txt](https://github.com/balena-os/balena-jetson-nano/files/3339402/log.do_configure.23539.txt)

acostach, 2019-06-28:
@floion this is a work in progress branch, until nvidia fixes the bootloader issues will do no PR on it.

@gancev @klutchell Have you tried with Ubuntu 18.04.2 LTS?

Does the bindiff have this md5sum on your side?
17500dc80a4da1d20929700a2bc6ad31  boot0.bindiff

Anyhow, I'll probably patch the binary directly with dd to avoid using bsdiff anymore.

klutchell, 2019-06-28:
Yeah, I'm also using Ubuntu 18.4 LTS. My MD5 sum also matches yours. I included the MD5 for the bspatch binary as well.

```
ecc8b8ea1d6d3798a5e83fb8c3efca51  ./balena-jetson-nano/tx2build/tmp/work/aarch64-poky-linux/tegra186-flash-dry/32.1.0-r0/recipe-sysroot-native/usr/bin/bspatch
17500dc80a4da1d20929700a2bc6ad31  ./balena-jetson-nano/layers/meta-balena-jetson/recipes-bsp/tegra-binaries/files/boot0.bindiff
```

acostach, 2019-06-28:
@klutchell Mine has a different md5sum, could be the root cause.
392ed8023436d2816e76d66d75db4373  bspatch

We'll try to avoid bsdiff altogether and will update the branch the next days.

gancev, 2019-06-28:
@klutchell and @acostach  ,
here are md5sum on my side.
~/balena-jetson-nano$ md5sum newboot.img boot_gen.img boot0.bindiff 
a81279ef6a7014e304bc58b5e96b1c00  newboot.img
b874e1990240a5370531d16a55d1b34e  boot0.img
17500dc80a4da1d20929700a2bc6ad31  boot0.bindiff

@klutchell  although bsdiff   is showing error,  boot0.img is generated and is valid ... at least on my side.
So I made small mods in the **tegra186-flash-dry_32.1.0.bb** to copy the image omitting the error.

The fastest way to verify on your side is if you have boot0.img on the disk  just comment this line in tegra186-flash-dry_32.1.0.bb:do_configure. 
>  #   bspatch newboot.img boot0.img ${WORKDIR}/boot0.bindiff

The build should finish with success  and you can flash the tegra tx2 with latest jetpack 4.2

acostach, 2019-07-01:
@gancev @klutchell 

I've updated the wip_warror_tx2 branch to the latest BalenaOS as well as removed bsdiff completely to switch to patching the image with dd.
```
$ git pull origin wip_warrior_tx2
$ git submodule update --init --recursive
$ cd buildtx2 && source ../layers/poky/oe-init-build-env . 
$ MACHINE=jetson-tx2 bitbake resin-image-flasher -c cleansstate && MACHINE=jetson-tx2 bitbake resin-image -c cleansstate
$ MACHINE=jetson-tx2 bitbake resin-image-flasher
```

klutchell, 2019-07-02:
Thanks for this change, the build is definitely progressing without bspatch failures!

However, now I'm running into a space calculation issue during do_rootfs. I'll try a fresh build on another machine to see if I get a different result, and if necessary I'll open a different issue.

```
NOTE: resin-boot: FAT overhead 7948 sectors, data 77308 sectors, largest file 77242 sectors, boot size 161920 sectors.
ERROR: resin-image-flasher-1.0-r0 do_rootfs: resin-boot: Not enough space for atomic copy operations.
NOTE: resin-boot: FAT overhead 7948 sectors, data 77308 sectors, largest file 77242 sectors, boot size 161920 sectors.
ERROR: resin-image-1.0-r0 do_rootfs: resin-boot: Not enough space for atomic copy operations.
```

acostach, 2019-07-03:
@klutchell I think it would be best to keep issues regarding this work in progress branch in this ticket, and open new tickets only for what is merged and deployed already, 

Is this a clean build? You're getting this error when files larger than half the size of the boot partition get copied to it. I've switched some configs to be built as modules so the kernel-initramfs image went down to around 38mb, so less than half the size of the boot partition, which is 80mb. On my side the images were generated well so I'm wondering your kernel may have some extra configs, patches? Was the kernel rebuilt and both resin-image* had a cleansstate?

klutchell, 2019-07-03:
Yeah, I ran cleansstate for both resin-image* steps and I also tried a fresh workspace on a different PC, still running into the same error. I can confirm my kernel-initramfs is also about 38mb. We have no modifications to the kernel on our side yet.

We aren't off by that much, but based on the sanity check I am getting the following sizes:
`FAT overhead 7940 sectors, data 77436 sectors, largest file 77370 sectors, boot size 161920 sectors`

I think the sanity check is failing because we are including the kernel size as part of the `DATA_SECTORS` calculation in `image-resin.bbclass:224` and then again in the `LARGEST_FILE_SECTORS` calculation in `image-resin.bbclass/image-resin.bbclass:234`. I also noticed a small typo in the `FAT_BYTES` calculation (references `RESIN_BOOT_WOKDIR`) but I don't see how that would cause this issue.

```
	# This is a sanity check
	# When updating the hostOS we are using atomic operations for copying new
	# files in the boot partition. This requires twice the size of a file with
	# every copy operation. This means that the boot partition needs to have
	# available at least free space as much as the largest file deployed.
	# First Calculate size of the data
	DATA_SECTORS=$(expr $(du --apparent-size -ks ${RESIN_BOOT_WORKDIR} | cut -f 1) \* 2)
	# Calculate fs overhead
	DIR_BYTES=$(expr $(find ${RESIN_BOOT_WORKDIR} | tail -n +2 | wc -l) \* 32)
	DIR_BYTES=$(expr $DIR_BYTES + $(expr $(find ${RESIN_BOOT_WORKDIR} -type d | tail -n +2 | wc -l) \* 32))
	FAT_BYTES=$(expr $DATA_SECTORS \* 4)
	FAT_BYTES=$(expr $FAT_BYTES + $(expr $(find ${RESIN_BOOT_WOKDIR} -type d | tail -n +2 | wc -l) \* 4))
	DIR_SECTORS=$(expr $(expr $DIR_BYTES + 511) / 512)
	FAT_SECTORS=$(expr $(expr $FAT_BYTES + 511) / 512 \* 2)
	FAT_OVERHEAD_SECTORS=$(expr $DIR_SECTORS + $FAT_SECTORS)
	# Find the largest file and calculate the size in sectors
	LARGEST_FILE_SECTORS=$(expr $(find ${RESIN_BOOT_WORKDIR} -type f -exec du --apparent-size -k {} + | sort -n -r | head -n1 | cut -f1) \* 2)
	if [ -n "$LARGEST_FILE_SECTORS" ]; then
		TOTAL_SECTORS=$(expr $DATA_SECTORS \+ $FAT_OVERHEAD_SECTORS \+ $LARGEST_FILE_SECTORS)
		BOOT_SIZE_SECTORS=$(expr ${RESIN_BOOT_SIZE} \* 2)
		bbnote "resin-boot: FAT overhead $FAT_OVERHEAD_SECTORS sectors, data $DATA_SECTORS sectors, largest file $LARGEST_FILE_SECTORS sectors, boot size $BOOT_SIZE_SECTORS sectors."
		if [ $TOTAL_SECTORS -gt $BOOT_SIZE_SECTORS ]; then
			bbfatal "resin-boot: Not enough space for atomic copy operations."
		fi
	fi
```
Even if this sanity check were not correct, I tried removing the bbfatal case and still failed to copy files into the image due to memory errors. I'm going to revisit that error this morning.

klutchell, 2019-07-03:
I stand corrected, fixing the typo on line 229 of `image-resin.bbclass` resolved the sanity check, I see now how the math was significantly impacted.

floion, 2019-07-03:
@klutchell thanks for the PR in meta-balena. I also commented there, can you please redo the PR against the `development` branch? That's the workflow used for meta-balena

klutchell, 2019-07-05:
I also have a commit in my fork that addresses `Stat for u-boot-dtb.bin failed` error during `tegra186-flash-dry:do_compile`. It's nearly identical to the fix for https://github.com/balena-os/balena-jetson-tx2/issues/81 but ported to `tegra186-flash-dry_32.1.0.bb`

https://github.com/klutchell/balena-jetson-nano/commit/3b569844cbbd82dbca1c891887c9e633166af263

Is your team interested in a PR for this change in the `wip_warrior_tx2` branch?

floion, 2019-07-05:
@klutchell sure thing, you can actually push a new PR against the master branch

klutchell, 2019-07-05:
@floion that file doesn't exist in master but I suppose it may also apply to `tegra194-flash-dry_32.1.0.bb`?

https://github.com/balena-os/balena-jetson-nano/blob/c78164342438f8ea5bc55c28d8d5bb57210ca4e0/layers/meta-balena-jetson/recipes-bsp/tegra-binaries/tegra186-flash-dry_32.1.0.bb#L218-L221

https://github.com/balena-os/balena-jetson-nano/blob/917b3d754de709f1f83a3616a01247bd2562de95/layers/meta-balena-jetson/recipes-bsp/tegra-binaries/tegra194-flash-dry_32.1.0.bb#L237-L240

I suppose I'm not clear on the purpose of the `wip_warrior_tx2` branch and the main differences between it and master. My goal is a development BalenaOS running on Jetson TX2 with L4T 32.1 and I was directed to this branch by the forums.

floion, 2019-07-05:
The goal of the `wip_warrior_tx2` is to be merged to master but currently we cannot merge it because of some issues related to host OS updates. We are waiting for a fix from NVidia and when that's released then we'll release that new L4T

klutchell, 2019-07-05:
Thanks @floion, I think I'll submit separate PRs to address both tegra186 and tegra194 since they exist only in separate branches right now. Does that work for you?

floion, 2019-07-05:
@acostach did you solve this like @klutchell suggests or was the thing I'm thinking about for a different package?

klutchell, 2019-07-11:
Hey @floion, the issue fixed by @acostach further up this thread was to avoid some issues with `bspatch` by using dd to adjust the image manually. That fix is working well for me.

The issue I would like to patch upstream is the dependency on `u-boot-dtb.bin` that causes `tegra186-flash-dry:do_compile` to fail on a clean workspace.

```
| [   0.0665 ] Creating list of images to be signed
| [   0.0684 ] tegrahost_v2 --chip 0x18 --partitionlayout flash.xml.bin --list images_list.xml zerosbk
| [   0.0722 ] Stat for u-boot-dtb.bin failed
```

 It's the same as the issue described and fixed [here](https://github.com/balena-os/balena-jetson-tx2/issues/81) but it impacts different branches and filenames.

floion, 2019-07-12:
Hi @klutchell as per https://github.com/balena-os/balena-jetson-tx2/pull/91#issuecomment-505118882 this issue should be fixed as here: https://github.com/balena-os/balena-jetson-tx2/commit/651530c64aa6faabede3e5ce1154b8c73e977814

acostach, 2019-07-15:
HI @klutchell I've updated the wip branch with u-boot install dependency.

Also it looks like they fixed the broken BSP archive download, the one brought in by tegra-binaries that suddenly could not be extracted. Gave it a try just now and seems to work fine again.

klutchell, 2019-07-16:
Thanks for the update @acostach, I can confirm that the bzip2 package errors have been fixed by NVIDIA.

However, I still see a u-boot error during `tegra186-flash-dry-32.1.0-r0 do_configure` in a fresh workspace with your latest changes.

```
| [   0.0258 ] tegrasign_v2 --key None --getmode mode.txt
| [   0.0583 ] Assuming zero filled SBK key
| [   0.0585 ]
| [   0.0587 ] Generating RCM messages
| [   0.0670 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x18 0 --download rcm mb1_recovery_prod.bin 0 0
| [   0.0864 ] RCM 0 is saved as rcm_0.rcm
| [   0.0869 ] RCM 1 is saved as rcm_1.rcm
| [   0.0869 ] List of rcm files are saved in rcm_list.xml
| [   0.0869 ]
| [   0.0869 ] Signing RCM messages
| [   0.1128 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key
| [   0.1295 ] Assuming zero filled SBK key
| [   0.1473 ]
| [   0.1474 ] Copying signature to RCM mesages
| [   0.1624 ] tegrarcm_v2 --chip 0x18 0 --updatesig rcm_list_signed.xml
| [   0.1869 ]
| [   0.1870 ] Parsing partition layout
| [   0.1983 ] tegraparser_v2 --pt flash.xml.in.tmp
| [   0.2189 ]
| [   0.2190 ] Creating list of images to be signed
| [   0.2364 ] tegrahost_v2 --chip 0x18 0 --partitionlayout flash.xml.in.bin --list images_list.xml zerosbk
| [   0.2481 ] Stat for u-boot-dtb.bin failed
| [   1.1685 ]
```
The only way I've found to resolve it is by adding `do_compile[depends] += "u-boot-tegra:do_deploy"` to the dependencies, same as https://github.com/ceresimaging/resin-jetson-tx2/pull/1/commits/174d38c1f81235433f6041ec414214da24cf736f.

Was the `virtual/bootloader` dependency you added intended to solve the same issue? It looks like `u-boot-dtb.bin` is not copied into the deploy/images folder at the time when `tegra186-flash-dry do_compile` is running.

acostach, 2019-07-25:
thanks @klutchell, checked again now and indeed this is the right way to do it, I'll include the fix when I'll have the branch updated for l4t 32.2, probably in the next one or two weeks, after the meta-tegra layer gets updated - https://github.com/madisongh/meta-tegra/issues/155

klutchell, 2019-07-25:
Thanks for the update @acostach, I've subscribed to https://github.com/madisongh/meta-tegra/issues/155 since I'm also waiting for the 32.2 release! I'll also keep an eye on open Issues in case there's anything I can help with.

klutchell, 2019-08-16:
All of my outstanding issues have been resolved by #12

floion, 2019-08-16:
Can this issue be closed now?

klutchell, 2019-08-16:
@floion Yes, I believe it can though I'm not the original reporter of the issue.

floion, 2019-08-16:
Right. So @gancev can this issue be closed?

gancev, 2019-08-19:
@floion yes it can be closed.

closed 2019-08-19 by acostach