balena-os/balena-jetson #11: [origin/wip_warrior_tx2] bzip2 fails to extract JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2

klutchell, 2019-07-02:
This issue isn't directly related to the balena-jetson-nano project, but it is encountered during a normal jetson-tx2 build so I wanted to document my workaround and see if there were more elegant options.

It seems the [L4T Driver Package (BSP)](https://developer.nvidia.com/embedded/dlc/l4t-jetson-driver-package-32-1-JAX-TX2) provided by NVIDIA does not extract correctly with current versions of bzip2. This causes the bitbake build to fail.

```
$ wget https://developer.download.nvidia.com/embedded/L4T/r32_Release_v1.0/jax-tx2/BSP/JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2

$ sha256sum JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2 
e7ecaa32928484398d2492e6e9911262b4663e720b4867404d8fe1928fe483aa  JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2
```
```
$ tar xjvf JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2 
Linux_for_Tegra/
Linux_for_Tegra/nv_tegra/
Linux_for_Tegra/nv_tegra/LICENSE.weston
Linux_for_Tegra/nv_tegra/nvidia_drivers.tbz2

bzip2: Data integrity error when decompressing.
        Input file = (stdin), output file = (stdout)

It is possible that the compressed file(s) have become corrupted.
You can use the -tvv option to test integrity of such files.

You can use the `bzip2recover' program to attempt to recover
data from undamaged sections of corrupted files.

tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
```
```
$ bzip2 -d JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2 

bzip2: Data integrity error when decompressing.
        Input file = JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tbz2, output file = JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tar

It is possible that the compressed file(s) have become corrupted.
You can use the -tvv option to test integrity of such files.

You can use the `bzip2recover' program to attempt to recover
data from undamaged sections of corrupted files.

bzip2: Deleting output file JAX-TX2-Jetson_Linux_R32.1.0_aarch64.tar, if it exists.
```
```
$ tar --version
tar (GNU tar) 1.29

$ bzip2 --version
bzip2, a block-sorting file compressor.  Version 1.0.6, 6-Sept-2010.

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic
```

My current workaround is as follows:
1. manually decompress the archive on another PC with just tar & lbzip and not a full bzip2 binary
2. recompress the archive and check for integrity errors
3. host the new archive locally and patch tegra-binaries-32.1.0.inc to point to the new file and hash

Has anyone else encountered this? I imagine most people have this file cached so it doesn't get redownloaded on an existing workspace.

If anyone has a more elegant workaround I'd love to hear it. I will likely open a ticket with NVIDIA to see if they can resolve the issue with the uploaded packages.

acostach, 2019-07-02:
Hi @klutchell 

In another ticket you mentioned issues with signing binaries for tegra186-flash-dry. This means that tegra-binaries - which brings this L4T archive - had been downloaded and unpacked at that point.

But probably you tried out a new build now and the archive is downloaded, but can't be unpacked anymore. 

Many people have reported this exact same issue these days on the nvidia forums: https://devtalk.nvidia.com/default/topic/1056381/jetson-agx-xavier/jetpack-4-2-xavier-install-failed/2
and the current workaround presented there is to downgrade bzip2 and libbz2 on the host side. We'll see what the official response is and take the appropriate steps.

klutchell, 2019-07-02:
Thanks for the link to the NVIDIA forums issue, I was just in the process of searching for that. My workaround of hosting the package locally is working for new builds so I'll continue to do that until NVIDIA fixes the issue on their side.

closed 2019-07-02 by klutchell

acostach, 2019-07-02:
Also discussed here: https://devtalk.nvidia.com/default/topic/1048691/jetson-tx2/jetpack-4-2-mdash-l4t-r32-1-release-for-jetson-agx-xavier-jetson-tx2-and-jetson-nano/4 but no official reply yet. Worth keeping an eye on both threads.