balena-os/balena-jetson #32: Support L4T 32.3 with Jetson-IO tool

JeffJassky, 2020-01-08:
In L4T 32.3 Nvidia includes the Jetson-IO tool to easily configure the dev kit 40-pin header pin.

We'd love to have this so we can configure PWM and I2S on the GPIO.

Docs for the Jetson-IO tool here: https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fhw_setup_jetson_io.html%23wwpID0E0EE0HA

acostach, 2020-01-09:
Hi @JeffJassky 

For what device are you interested in running this? Is it the jetson-nano?

Not really sure how that tool works, if it needs an updated kernel, but I think you could first give it a shot by downloading and applying the 32.3 BSP archive in the container, on top of a L4T 32.2 based BalenaOS image.

JeffJassky, 2020-01-16:
Thanks, @acostach. Yes interested in having this for the nano.

Any chance you can provide resources on how to download/apply a BSP in the container?

Thanks!

acostach, 2020-01-16:
@JeffJassky , here's how I unpacking the 32.3 BSP archive would look like on the nano:

```
FROM balenalib/jetson-nano-ubuntu:bionic

ENV DEBIAN_FRONTEND noninteractive

WORKDIR /usr/src/app

RUN \
apt-get update && apt-get install --no-install-recommends -y lbzip2 wget tar && \
wget https://developer.nvidia.com/embedded/dlc/r32-3-1_Release_v1.0/t210ref_release_aarch64/Tegra210_Linux_R32.3.1_aarch64.tbz2 && \
tar -xf Tegra210_Linux_R32.3.1_aarch64.tbz2 && \
./Linux_for_Tegra/apply_binaries.sh -r / && \
rm Tegra210_Linux_R32.3.1_aarch64.tbz2 && \
echo "/usr/lib/aarch64-linux-gnu/tegra" > /etc/ld.so.conf.d/nvidia-tegra.conf && ldconfig

ENV UDEV=1

CMD ["bash"]
```

You can find the BSP link from above at https://developer.nvidia.com/embedded/linux-tegra

I skimmed trough the docs of that tool, it says it generates a new dtb and applies it trough the extlinux.conf file. So probably you will have copy that generated dtb from the container to the hostOS and apply the extlinux.conf changes in the hostOS (in /mnt/boot/extlinux/extlinux.conf). I know dtb selection from the extlinux.conf on L4T 28.x and prior releases was disabled from uboot, but it may have been added in 32.X. 

Let us know if you get this approach to work and maybe we can look into ways of integrating some of these pre-generated dtbs into the hostOS directly, to switch between them from the dashboard like it's done for the raspberry pi family of boards.

JeffJassky, 2020-01-23:
I'm told by Nvidia that the tool will not work on 32.2. Here's a thread from nvidia: https://devtalk.nvidia.com/default/topic/1069938/jetson-nano/jetson-io-tool-on-32-2-/post/5421999/#5421999

When might we expect the Balena images for jetson nano to support 32.3?

acostach, 2020-01-24:
@JeffJassky We don't have a timeline for this currently. But if the tool generates a new dtb, and uboot switches back to allowing loading of the dtb from extlinux.conf, then we need to analyze how and if this use-case can be accommodated, without introducing breaking changes.

I believe the quickest way to achieve this is to modify the dtb directly for your application needs.

acostach, 2020-01-27:
@JeffJassky I've investigated this use-case a bit and raised https://github.com/balena-os/balena-jetson/issues/37 to track progress on updating to 32.3 to include custom dtb loading. We cannot provide a timeline for the upgrade yet, but you keep an eye on that issue. 

Is it fine if we close this ticket?

acostach, 2020-02-06:
Closing this issue as we are tracking support for custom dtbs in #37

closed 2020-02-06 by acostach

acostach, 2020-03-09:
The 2.47.1+rev3 release for the Nano has support L4T 32.3.1, so it now supports custom dtb loading. This means that in the hostOS, in /mnt/sysroot/active/current/boot/extlinux/extlinux.conf you can now specify your own FDT:
...
`MENU LABEL primary Image`
`LINUX /Image`
`FDT /boot/<your_dtb_name>` <---
`APPEND .....`

and you can place the already compiled DTB in the hostOS, inside /mnt/sysroot/active/current/boot/
You can't run jetson-gpio in the hostOS, but you can build and save a couple DTBs on a generic ubuntu image and then copy them to the hostOS.

AeroNotix, 2020-05-26:
@acostach how would we go about doing that from the application/container-side?

acostach, 2020-05-26:
@AeroNotix container is isolated, and can't write to the OS. You can place the file manually in the hostOS for testing. Please note that the dtb referenced in the extlinux.conf needs to be generated by the jetson-gpio tool, other dtbs didn't boot when I tested on the Nano.