balena-os/balena-jetson #61: Support for Jetson Xavier NX

maxritter, 2020-04-27:
Would be great to have the new xavier nx supported as well! Jetpack 4.4 (developer edition) is out that has the drivers ;)

acostach, 2020-04-30:
Hi Max, looks like the meta-tegra layer does not yet support the NX machine, but there is a ticket there stating it will be added soon. 

Are you using the NX with the B01 Nano devkit carrier board? Or maybe know if it's fully compatible, apart from the known power limitations?

maxritter, 2020-04-30:
Thanks for letting me know! So we will wait for the meta-tegra update then :)

We are waiting for an answer from Auvidea concerning the best option for a NX carrier board. Probably will be the JNX30 board. But yeah, the B01 should support the NX module as well, as long as it does not draw too much current I think.

maxritter, 2020-05-04:
@acostach  Meta-Tegra was updated recently and now has experimental support for Jetson Xavier NX eMMC module in Nano carrier board. Is this good enough to get us going?

acostach, 2020-05-04:
@maxritter I've asked Matt from meta-tegra in the nx ticket to confirm his setup so we can order the same hardware and investigate support for the NX.

maxritter, 2020-05-04:
Great, looking forward to it!

dremsol, 2020-05-18:
Probably you are aware but NVIDIA dropped a [NX devkit](https://developer.nvidia.com/embedded/jetson-xavier-nx-devkit), didn't see that one coming... Anyway, we will receive one tomorrow and start some testing.

acostach, 2020-05-21:
Yes, we received the module and B01 nano board and started investigating it.

Currently we encountered some issues caused by cboot, we will address them in https://github.com/balena-os/meta-balena/issues/1905 and will keep you updated here on the progress.

dremsol, 2020-05-28:
hi @acostach, what kind of NX device you will be supporting? The official NX devkit or NX eMMC module in dev kit or Nano carrier board?

We have everything in-house to contribute the `photon-xavier-nx` device which is based on the eMMC module. Do you have a suggestion on which machine to base upon?

dremsol, 2020-06-04:
@acostach we have preliminary support for photon-xavier-nx. We will hold with a PR until following is merged;
https://github.com/balena-os/balena-jetson/pull/74

acostach, 2020-06-04:
@dremsol perfect. I assume you were able to build on top of the nx machine in this PR, right?

dremsol, 2020-06-04:
That's correct. Would be nice if we can pull it in once both https://github.com/balena-os/balena-jetson/pull/74 and https://github.com/balena-os/balena-jetson/pull/75 are merged.

JohnBetaCode, 2020-07-30:
when it is estimated to have support for Xavier NX Jetpack 4.4?

acostach, 2020-10-06:
@JohnBetaCode an image with L4T 32.4.3 is now available for testing in the [staging dashboard](https://dashboard.balena-staging.com), it's the image version 2.58.3+rev3.

Can be flashed on the target using: https://github.com/balena-os/jetson-flash/pull/30

Closing ticket as support for the NX is available in production dashboard.

closed 2020-10-06 by acostach

compiaffe, 2020-10-12:
@acostach Apologies for raising a closed issue, but does the Xavier NX image also support the SD card version (aka devkit?).

acostach, 2020-10-12:
Hello @compiaffe,

No, this image only works for the NX emmc version.

compiaffe, 2020-10-12:
Thanks @acostach. Any plans to support the devkit and if yes, any place I can keep myself up to date?

acostach, 2020-10-12:
@compiaffe we don't have this particular version of the hardware so for the moment it's not on the list, but will let you know if we have an update

acostach, 2020-10-19:
@compiaffe the nx devkit (sd-card) is now on our list. Planning to start looking into it in the next few weeks, when the hardware arrives.

acostach, 2020-11-02:
@compiaffe the NX Devkit SD-CARD image is now available for testing at https://dashboard.balena-staging.com/