balena-os/balena-jetson #66: [jetson-nano]: B01 revision runs slower with A02 dtb

acostach, 2020-05-18:
Currently we're using in our images the a02 revision dtb. A newer jetson-nano devkit with sd-card is now sold and is at revision b01.

b01 performs slower with a balena images that has the a02 dtb, like for instance 2.47.1+rev3 and this application: https://github.com/pascal-hwky/balenaos-slowdown , even in maxn mode. With the b00 dtb both boards show good results in speed:
```

18.05.20 13:23:37 (+0300)  main  Finished 100 inferences.
18.05.20 13:23:37 (+0300)  main  Max time: 147 [ms]
18.05.20 13:23:37 (+0300)  main  Min time: 91 [ms]
18.05.20 13:23:37 (+0300)  main  Avg time: 105 [ms]
```

According to nvidia engineers at https://forums.developer.nvidia.com/t/part-number-confusion-with-jetson-nano-and-create-jetson-nano-sd-card-image-sh/107369 

> "B0x images will work on A02 boards without issue. There are some pinmux differences between the two hardware revisions, but the B0x pinmux settings will work fine on an A02 board.".

I could test on the A02 that hdmi, usb and ethernet work fine with <pre>tegra210-p3448-0000-p3449-0000-b00.dtb.encrypt</pre>

In this context, we can look further into switching to this dtb in the next release.

acostach, 2020-06-17:
Fixed in production 2.51.1

closed 2020-06-17 by acostach

georgiats, 2020-06-22:
@acostach this user came back and reported that now with 2.51.1+rev1 everything seems snappy again. The only thing they've noticed is that the average inference time is still somewhat slower, but as far as I understand this is not something we have to worry about. 

```balenaOS 2.45.1+rev3 (production), supervisor 10.3.7

Finished 100 inferences.
Max time: 149 [ms]
Min time: 75 [ms]
Avg time: 78 [ms]

balenaOS 2.51.1+rev1 (production), supervisor 11.4.10

Finished 100 inferences.
Max time: 146 [ms]
Min time: 75 [ms]
Avg time: 103 [ms]```