balena-os/balena-jetson #68: [jetson-xavier] Flashing more recent images

neptune-files, 2020-05-26:
I have been tracking the updates to the balena-jetson branch and would like to run the later builds that incorporate newer jetpack versions. 

The downloaded image from balena cloud v2.43.0+rev4 installs using https://github.com/balena-os/jetson-flash. I can successfully build much later tagged versions checking out different tags and then running `balena-yocto-scripts/build/barys -m jetson-xavier -d`

I am interested in flashing a 32.3.1 image as we need features from that version.

To flash a  32.3.1 I have updated the jetson-flash url to use a 32.3.1 BSP pack as described in by @acostach in https://github.com/balena-os/jetson-flash/issues/22#issuecomment-595248247

after making that update and attempting to flash using the following command `./bin/cmd.js -f ../balena-jetson/build/tmp/deploy/images/jetson-xavier/resin-image-jetson-xavier.resinos-img -m jetson-xavier` 

The flash succeeds and the xavier boots the kernel never brings up balenaOS. It hangs saying `random: crng init done` which I assume is not related just the last init before it would boot into the right partition.

I have also tried with and without updating https://github.com/balena-os/jetson-flash/blob/master/assets/jetson-xavier-assets/resinOS-flash.xml to the resinOS-flash xml from the relevant release.

Strangely with the updated 32.3.1 BSP jetson-flash will successfully flash the downloaded v2.43.0+rev4 image.
  
Any help flashing built images would be great.

Related is it possible to download these later releases from the balena staging server (or production)?

`BALENARC_BALENA_URL=balena-staging.com balena os download jetson-xavier --output img.img --version menu`  does not offer anything later than v2.43.0+rev4 the CI seems to be building later versions do they live anywhere? 

Either on the staging or non-staging server. It would allow me to sanity check that it is not an issue with later version incompatibility with the xavier?

acostach, 2020-05-26:
@neptune-files For this specific L4T release 32.3.1 you could try build the v2.48.0+rev1 tag. Please note that images become available in the dashboard only after they are tested for the updated L4T. For the Xavier, we are currently investigating L4T 32.4.2 to bring to the dashboard.

neptune-files, 2020-05-26:
Thanks! I successfully installed v2.50.1+rev1 the L4T 32.4.2 image from your latest commits in the master branch. That will work for us. I appreciate your help. I look forward to it coming to the dashboard and I'll keep an eye out.

closed 2020-05-26 by neptune-files