balena-os/balena-jetson #72: Update to L4T 32.4.2

acostach, 2020-05-29:
Hello @compiaffe, @dremsol, @sjdrc,

We're currently working on updating the jetsons to L4T 32.4.2. When updated dtbs will become available from the carrier board manufacturers, please feel free to PR them for the community boards you are maintaining. 

Thanks

dremsol, 2020-05-29:
Hi @acostach,

I've been contacting CTI for their l4t 32.4.2 BSP releases for following devices;
- photon-nano (not yet available)
- photon-xavier-nx (we have the BSP sources but NX not yet supported in balena-jetson)

We have decided to move away from Auvidea as we found the quality of their boards don't reach our standards. However, I will take a look to their firmware time to time to see if their is a update for l4t 32.4.2 relating;
- [jn30b-nano](https://auvidea.eu/firmware/)

acostach, 2020-05-29:
Perfect, thank you for the update @dremsol. Will let you know as soon as there's something for the nx, we're currently working on it.

compiaffe, 2020-06-19:
@acostach I've reached out to Aetina to ask. In my experience they skip a few versions and their old DTB might still work for newer versions. Last time NVIDIA changed the partitions it did require a new DTB of course. Will keep you posted.

dremsol, 2020-07-14:
Hi @acostach, by merging #82 all our devices will be updated to l4t-32.3.2. Once you merge #75 could these devices be synced with this particular OS version?

compiaffe, 2020-07-16:
@acostach FYI: No news on 32.4.2 / Jetpack 4.4 from Aetina yet.

acostach, 2020-07-16:
thank you for the update @compiaffe !

@dremsol to merge #75 my colleagues on the OS side are currently working on a solution to include the libs necessary for the nvidia hook to run. 

The photon-nx is already available in staging dashboard, and yes, of course, we will release new production images for all of them to include that PR once the space problem is solved.

compiaffe, 2020-07-20:
@acostach FYI: I've got the dtb for N510 - ETA of MR: End of week/beginning of next

dremsol, 2020-08-03:
@acostach in the meantime we have acquired the `l4t-32.4.3` BSP sources for both `photon-nano` and `photon-nx`. We will make a PR for both of them. Any indication when master tracks `l4t-32.4.3`?

pgils, 2020-10-01:
Thanks @acostach ! I've just opened PR #107 for photon-*.

acostach, 2020-10-02:
Thanks @pgils , I merged #107.

Please note that hostapp-extensions from meta-balena is currently under development, and a release to production to include it will take a while due to the increased complexity of this feature. if you need these two boards in production without the hostapp extensions, I can first release them in the staging dashboard so you can check the hostOS update and what else you find necessary.

compiaffe, 2020-10-07:
@acostach The newest dtb (32.4.2) was already merged as per #94 / #95 
Thanks for the nudge.

acostach, 2020-11-02:
Closing for now, will re-open when DTBs for community supported boards need to be updated.

closed 2020-11-02 by acostach