balena-os/balena-jetson #104: Support for Rogue AGX 101 Carrier board

jonathankehler, 2020-09-23:
Stemmed from https://forums.balena.io/t/support-connect-tech-rogue-agx-carrier-board-for-jetson-xavier/192168

Looking for help creating a base image for the Rogue AGX 101 carrier board for Jetson Xavier AGX.
https://connecttech.com/product/rogue-carrier-nvidia-jetson-agx-xavier/

I'm attaching the dtb from the bsp Connect-tech (CTI) provides as a download:
https://connecttech.com/ftp/Drivers/CTI-L4T-AGX-32.3.1-V007.tgz

Also providing the dtb we created that includes our required sensor support.

[dtb-files.zip](https://github.com/balena-os/balena-jetson/files/5270545/dtb-files.zip)

jonathankehler, 2020-09-23:
Just to be explicit we are trying to build a base image based on L4T 32.3.1 as referenced here https://forums.balena.io/t/balena-jetson-image-for-l4t-r32-3-1/180760/12

acostach, 2020-09-24:
Hi @jonathankehler 

Can you please open a PR so we can see the exact changes that you did for your custom build?

acostach, 2020-09-28:
@jonathankehler I added some changes for the Rogue, try building this branch: https://github.com/balena-os/balena-jetson/commits/wip_rogue_xavier_agx

You will need to do a git submodule update --init --recursive after the branch checkout

jonathankehler, 2020-09-28:
Thank you @acostach.
I will try these out.

The problem we are facing now is that your branch is based off of L4T 32.4.2 which is a non-production developer preview.
I see that you are working on 32.4.3 support - can we get your rogue changes + the 32.4.3 support?

The board support files we have from Connect-tech (who make the Rogue) are based on 32.4.3 (or 32.3.1 alternatively).

acostach, 2020-09-29:
Hi @jonathankehler ,

I see. Yes, I'll port the Rogue changes to my 32.4.3 WIP branch.

acostach, 2020-09-29:
Hi @jonathankehler , I added the cti rogue to the 32.4.3 wip branch, you can do a local build using the branch in this PR https://github.com/balena-os/balena-jetson/pull/105

`./balena-yocto-scripts/build/barys -b build_rogue -m cti-rogue-xavier`

After the image is generated, please use this [PR](https://github.com/balena-os/jetson-flash/pull/30) to flash the image to the board: 
`
./bin/cmd.js -m jetson-xavier -f resin-image-cti-rogue-xavier-20200929xxxx.rootfs.resinos-img`

acostach, 2020-12-23:
Closing due to inactivity, feel free to re-open if necessary.

closed 2020-12-23 by acostach

acostach, 2021-02-12:
Hi @jonathankehler do you still use this machine for local builds, or should we remove it?

jonathankehler, 2021-02-22:
Yes we are continuing to use this machine so keeping support would be helpful. Thank you!

acostach, 2021-02-22:
Hi @jonathankehler, we can make it available as a community device at https://dashboard.balena-cloud.com, supported by you. If you wish to have the image available there, please open a PR to update the DTB to 32.4.4 and any other custom changes you are using.

acostach, 2021-05-17:
Hi @jonathankehler , in case you missed my ping above and want to have this device type available as a community board supported by you, please PR a L4T 32.4.4 dtb and we can make it available as a pre-built image in the dashboard.