balena-os/balena-jetson #106: Support for NRU-120s Carrier board

tekollt, 2020-09-28:
We would like to have support for Nvidia Jetson Xavier AGX with custom DTB and neousys bard.
Can we use variable 'RESIN_HOST_EXTLINUX_fdt' or some way else to specify the custom device tree? 

The DTB provided to us by the provider is supported for L4T 32.4.3

acostach, 2020-09-28:
Hi @tekollt , currently we're only supporting the Devkit for the AGX, but you can try a local build with your dtb using the https://github.com/balena-os/balena-jetson/tree/wip_rogue_xavier_agx branch. It is based on 32.4.2 so the 32.4.3 dtb should work.

The Xavier devices don't use u-boot so the kernel and dtb are read from raw partitions. To do the local build you will need to:

- add your device tree in linux-tegra - https://github.com/balena-os/balena-jetson/commit/d1ce3d865505459b899caecb53dfa957b9157cba
- add the yocto machine - https://github.com/balena-os/balena-jetson/commit/b3b8444ebdb2e9bcc92ed11c3f8f676543aaf03e 
- mention the DTBNAME in https://github.com/balena-os/balena-jetson/commit/411cc9b38d08a0c905eb5d0715e9a493932e9030#diff-20bf8880bcf441eca476788145557015R17 
- add a coffee file: https://github.com/balena-os/balena-jetson/commit/853a259d4c5bdf548b08382c481a630a7108123e

To start a local build:
./balena-yocto-scripts/build/barys -b <build_dir> -m <machine_name>

To flash the resulting image you can use https://github.com/balena-os/jetson-flash

tekollt, 2020-09-28:
It is a bit unclear how to proceed with this, do there exist some more detailed manual of how to make this work?

Would we be able to take our current dtb file and inect into the img file we download from balena, or manually test this before we go all the way to create the custom image?

acostach, 2020-09-28:
Hi @tekollt,

It's not possible to inject the dtb file for the Xavier devices as the vendor does not provide u-boot support.

You can upload the DTB here and I can update the branch with it, then you can then just do the build locally with barrys and flash the resulting image on your device for testing.

tekollt, 2020-09-28:
That would be great, and really apriciated.
Do you operate with a certain quantity to prepare a dedicated image?

Last ned Outlook for Android<https://aka.ms/ghei36>

tekollt, 2020-09-28:
That would be great, and really apriciated.
Will try to get the image uploaded tomorrow CEST. 

Do you operate with a certain quantity to prepare a dedicated image?

acostach, 2020-09-29:
Hi @tekollt , we'll need only the board's DTB, an entire image does not help.

I will update the branch to include the DTB you provide and then you can build the BalenaOS image locally on your PC.

tekollt, 2020-09-29:
I have made the DTB avaliable from this location @acostach  https://drive.google.com/file/d/1OvOTW2GNOzmpKrRHCzrrzja-j0mhsPlX/view?usp=sharing

acostach, 2020-09-29:
Hi @tekollt, please use this PR for your local build: #105 . After checking out the branch, you'll need to to a `git submodule update --init --recursive`

To start the build:
`./balena-yocto-scripts/build/barys -b build_nru120s -m nru120s-xavier`

After the image is generated, please use https://github.com/balena-os/jetson-flash/pull/30 to flash the image to the board:
`./bin/cmd.js -m jetson-xavier -f resin-image-nru120s-xavier-xxxxx.rootfs.resinos-img`

tekollt, 2020-09-30:
I am getting some checksum errors during the build, is this something i should worry about @acostach  ? 
And are there hardware utilization that would allow me to speed up the build process? running for 4th our and only 49%

 
```
Sstate summary: Wanted 1411 Found 0 Missed 1411 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: popt-native-1.16-r3 do_fetch: Checksum mismatch for local file /home/user/balena-jetson/build_nru120s/downloads/popt-1.16.tar.gz
Cleaning and trying again.
WARNING: popt-native-1.16-r3 do_fetch: Renaming /home/user/balena-jetson/build_nru120s/downloads/popt-1.16.tar.gz to /home/user/balena-jetson/build_nru120s/downloads/popt-1.16.tar.gz_bad-checksum_9e09d0bf772649c5ba385c48ba2d3547
WARNING: popt-native-1.16-r3 do_fetch: Checksum failure encountered with download of http://rpm5.org/files/popt/popt-1.16.tar.gz - will attempt other sources if available

WARNING: icu-native-63.1-r0 do_fetch: Checksum mismatch for local file /home/user/balena-jetson/build_nru120s/downloads/icu4c-63_1-src.tgz
Cleaning and trying again.
WARNING: icu-native-63.1-r0 do_fetch: Renaming /home/user/balena-jetson/build_nru120s/downloads/icu4c-63_1-src.tgz to /home/user/balena-jetson/build_nru120s/downloads/icu4c-63_1-src.tgz_bad-checksum_8acd6f1a693270f216a94ea3f11826fa
WARNING: icu-native-63.1-r0 do_fetch: Checksum failure encountered with download of http://download.icu-project.org/files/icu4c/63.1/icu4c-63_1-src.tgz - will attempt other sources if available
Currently  4 running tasks (1921 of 3934)  48% 

```

acostach, 2020-09-30:
@tekollt these appear to be warnings from the logs you posted, like the download failed and the build system is retrying. Probably some network issues there.

Yes, Yocto builds usually take a bit of time, depending on the machine you're running them on, on your network speed, etc.

Also, for yocto builds in general it's recommended to use shared-sstate and shared-downloads so that new builds take way less time - https://github.com/balena-os/balena-yocto-scripts/blob/13d998ce54d12b71fa5d96035dee11c03436c914/build/barys#L91

tekollt, 2020-10-01:
@acostach We managed to build the image. Dont know how i have overlooked to ask for that previously, but how can we assign the image beeing built to an appliation in our dashboard ?

acostach, 2020-10-01:
@tekollt before flashing the image with the jetson-flash PR I pasted above, you can losetup -fP the image you built, and then use https://www.balena.io/docs/reference/balena-cli/#config-inject-file

For instance:
```
sudo losetup -fP resin-image-nru120s-tx2-..rootfs.resinos-img
dmesg | tail <- check what loop device was created
balena config inject ./myconfig.json --type jetson-xavier --drive /dev/loopXY
sudo losetup -d /dev/loopXY
sudo losetup -D
```

tekollt, 2020-10-01:
Great, thanks! 
Is it possible for us to inject some additional Wifi drivers the same way? we have a jj jws6058 PCIE card attached which we are not able to use for the moment. Or should we try to use the broadcome based cards?

acostach, 2020-10-01:
@tekollt no, that's not possible. You can patch the kernel though to port the necessary driver if not present in nvidia kernel, and then build it as a module from the configs in linux-tegra_%.bbappend. Or build the driver for your wifi module in container, and load it from there with insmod.

tekollt, 2020-11-02:
@acostach  would we have to do anything extra to allow for GPU/Cuda support to a multi-container setup?
I found some references to io.balena.features.gpu but not sure if that is applicable

acostach, 2020-11-02:
@tekollt In the app's [docker compose](https://github.com/balena-io-examples/tx2-container-contracts-sample/blob/6efd05f2b023f6dd338fbf11a96b9b79e043307a/docker-compose.yml#L17) file you will need to add "privileged: true", and [enable udev](https://github.com/balena-io-examples/tx2-container-contracts-sample/blob/6efd05f2b023f6dd338fbf11a96b9b79e043307a/tx2_32_4_2/Dockerfile.template#L45) in container. [Base images ](https://github.com/balena-io-library/base-images/blob/91228b8f7785c025ac662ff38d42b119364ccdfd/balena-base-images/device-base/jetson-xavier/ubuntu/bionic/run/Dockerfile#L4) already have the repositories added so all you need to do is [unpack and apply](https://github.com/balena-io-examples/tx2-container-contracts-sample/blob/6efd05f2b023f6dd338fbf11a96b9b79e043307a/tx2_32_4_2/Dockerfile.template#L17) the BSP archive in container, apt-get update && apt-get install cuda packages.

tekollt, 2020-11-02:
Tried running from base balenalib/jetson-tx2-ubuntu:bionic but where not able to install cuda with "apt install cuda" even if the repo`s are added in your base. 

But running with privileged:true and enable udev, should that also work if we chose use for example the nvidia image nvcr.io/nvidia/l4t-tensorflow:r32.4.3-tf1.15-py3  ?

acostach, 2020-11-03:
@tekollt for the xavier you'll need to use a xavier base image. Here's a trimmed down Dockerfile based on the one I linked above:

```
FROM balenalib/jetson-xavier-ubuntu:bionic

# Prevent apt-get prompting for input
ENV DEBIAN_FRONTEND noninteractive

# Download and install BSP binaries for L4T 32.4.2
RUN \
  apt-get update && apt-get install -y --no-install-recommends lbzip2 wget tar && \
  cd /tmp/ && wget https://developer.nvidia.com/embedded/L4T/r32_Release_v4.2/t186ref_release_aarch64/Tegra186_Linux_R32.4.2_aarch64.tbz2 && \
  tar xf Tegra186_Linux_R32.4.2_aarch64.tbz2 && rm Tegra186_Linux_R32.4.2_aarch64.tbz2 && \
  cd Linux_for_Tegra && \
  sed -i 's/config.tbz2\"/config.tbz2\" --exclude=etc\/hosts --exclude=etc\/hostname/g' apply_binaries.sh && \
  sed -i 's/install --owner=root --group=root \"${QEMU_BIN}\" \"${L4T_ROOTFS_DIR}\/usr\/bin\/\"/#install --owner=root --group=root \"${QEMU_BIN}\" \"${L4T_ROOTFS_DIR}\/usr\/bin\/\"/g' nv_tegra/nv-apply-debs.sh && \
  sed -i 's/LC_ALL=C chroot . mount -t proc none \/proc/ /g' nv_tegra/nv-apply-debs.sh && \
  sed -i 's/umount ${L4T_ROOTFS_DIR}\/proc/ /g' nv_tegra/nv-apply-debs.sh && \
  sed -i 's/chroot . \//  /g' nv_tegra/nv-apply-debs.sh && \
  ./apply_binaries.sh -r / --target-overlay && cd .. \
  rm -rf Linux_for_Tegra && \
  echo "/usr/lib/aarch64-linux-gnu/tegra" > /etc/ld.so.conf.d/nvidia-tegra.conf && ldconfig

# Install gstreamer, X and xfce
RUN \
  apt-get install -y --no-install-recommends \
  xserver-xorg-input-evdev \
  xinit \
  xfce4 \
  xfce4-terminal \
  x11-xserver-utils \
  dbus-x11 \
  matchbox-keyboard \
  xterm 

ENV XFCE_PANEL_MIGRATE_DEFAULT=1
ENV UDEV=1

# Prevent screen from turning off
RUN echo "#!/bin/bash" > /etc/X11/xinit/xserverrc \
  && echo "" >> /etc/X11/xinit/xserverrc \
  && echo 'exec /usr/bin/X -s 0 dpms' >> /etc/X11/xinit/xserverrc

# Start XFCE desktop
CMD ["bash"]
```

Then, in the dashboard's container web-terminal you can do:

```
apt-cache search cuda
apt-get install -y cuda-samples-10-0
cd /usr/local/cuda-10.0/samples/2_Graphics/marchingCubes/
make
X&
```

Then run the cuda graphic example to see the animation:
```
root@8267885:/usr/local/cuda-10.0/samples/2_Graphics/marchingCubes# ./marchingCubes 
[./marchingCubes] - Starting...
MarchingCubes
GPU Device 0: "Xavier" with compute capability 7.2

grid: 32 x 32 x 32 = 32768 voxels
max verts = 102400
Read './data/Bucky.raw', 32768 bytes
```

The nvcr base images cannot be used right now, support is on our roadmap but we cannot offer a timeline. You can install all the packages you need in your Application containers, as shown in the above example.

closed 2020-11-16 by tekollt

acostach, 2021-02-12:
Hi @tekollt are you still using this machine to do local builds or can we remove it?

tekollt, 2021-02-12:
@acostach  yes we are, we are actively using this image on the devices in our fleet.
Could we somehow simplyfi things and get an official image downloadable from the website instead?

acostach, 2021-02-12:
Hi @tekollt , sure. Wasn't aware since you closed the ticket a while ago without letting us know that you would like to continue to use this custom community board.

acostach, 2021-02-12:
So are you currently using your custom build with some additional kernel patches or changes? Currently the repository is based on L4T 32.4.4, so if you have a L4T 32.4.4 dtb or other changes, please PR them here so we can include them.

acostach, 2021-05-17:
Hi @tekollt, just wanted to remind you that if you want to have this image available in the dashboard as a community board supported by you, please PR the L4T 32.4.4 dtb and we can publish it.