balena-os/balena-jetson #109: Provide a CUDA hostapp extension so that CUDA applications can run

alexgg, 2020-10-05:
Once the hostapp extension support in https://github.com/balena-os/meta-balena/pull/1987 is merged, provide a CUDA specific hostapp extension to include CUDA libraries that cannot be placed on a user application container.

This will allow CUDA applications to be run.

Also relates to https://github.com/balena-io/balena-supervisor/issues/1449

acostach, 2020-10-05:
Note: Will need to ensure the libraries in the extension match the L4T version in the OS, as we've just updated master branch from 32.4.2 to 32.4.3 for the following boards: Nano SD-CARD, Nano eMMC, TX2, Xavier AGX, Xavier NX Devkit eMMC, Photon Xavier NX and Photon Nano

klvnptr, 2020-10-22:
Can you guys please provide a "Xavier NX Devkit" image without the eMMC? We currently bought two Xavier NX Devkits from Arrow to get started with Balena development, and we realised it only supports old eMMC Devkits, which cannot be bought anymore. Thank you.

acostach, 2020-10-22:
Hi @klvnptr, the NX Devkit is on our list to add BalenaOS support.

This ticket is about CUDA host extensions, we will post updates about the SD-CARD version in the Xavier NX ticket here: https://github.com/balena-os/balena-jetson/issues/61#issuecomment-712203865

klvnptr, 2020-11-08:
@acostach something is not okay iwth the new SD card NX image. update won't download. keeps restarting. i grantend support access to this device: https://dashboard.balena-staging.com/devices/0070a4bda80d546dfc50956d9b5a399e/summary

klvnptr, 2020-11-23:
@acostach may I ask, how is the CUDA hostapp extension coming along? when can we expect some beta release? :) thank you.

acostach, 2020-11-23:
@klvnptr my colleagues are actively working on developing this feature, we cannot provide a timeline but will let you know when it becomes available for testing in the staging dashboard.

klvnptr, 2020-11-23:
@acostach thanks. at least may I get some estimation, when will you ship this features in staging? 1 month? 3 month? 1 year? because depending on your answer, we might need to take another route how we inject drivers into balena OS. thank you.

alexgg, 2020-12-04:
Connects-to: https://github.com/balena-os/meta-balena/pull/2052