balena-os/balena-jetson #120: unrealisticly high memory usage for multicontainer udp port forwarding

klvnptr, 2020-11-07:
HOST OS VERSION: balenaOS 2.56.0+rev1 production
SUPERVISOR VERSION: 11.14.0

way to reproduce the problem:
1. create a docker-compose.yml with the following content:
```yaml
version: '2.1'

services:
  test0:
    container_name: test0
    build: ./test0
    # network_mode: host
    ports:
      - "5004"
      - "10000-10200/udp"
      - "8088"
      - "8188"
      - "1022:1022"
  test1:
    container_name: test1
    build: ./test1
    ports:
      - "22:22"
      - "1234:1234"
      - "5000:5000"
```
2. now create two empty folders `test0` and `test1` and create a Dockerfile in both of them with the following content:
```
FROM balenalib/jetson-nano-ubuntu:bionic-run
CMD ["sleep", "infinity"]
```
3. deploy this multicontainer app to a jetson nano sdcard devkit
```
$ balena deploy app_name --build
```
4. watch memory usage of the host OS going through the roof. using about 3.5Gb. it can be checked 
5. now in the docker-compose.yml uncomment the `network_mode: host` line, and comment out all ports. so basically it turns off balena-engine-proxy, that does the port forwarding for test0 container.
6. deploy it. like in step 4. and watch that memory usage is still high. 
7. now reboot the device from balena cloud, wait for both containers to start. memory usage is minimal as it should be.

please note that it doesn't have to be rebooted to fix memory usage, if you never ever deploy test0 container with the 5 port forwarding lines uncommented in step 1. only use network_mode: host, then everything is okay. probably something gets screwed up when doing port forwarding for test0, and even a redeploy without won't fix it.

acostach, 2020-11-09:
Hi,

From the description you posted this doesn't seem to be something specific to jetson-nano or nvidia boards, sounds more like a process is started by the engine for each port, see https://forums.balena.io/t/large-number-of-forwarded-udp-ports-break-the-device/36364

acostach, 2020-11-20:
Not a bug in the balena-jetson repository, closing for now due to lack of activity. Feel free to re-open if necessary.

closed 2020-11-20 by acostach