balena-os/balena-jetson #122: Support for the Jetson Nano 2GB developer kit

alexgg, 2020-11-12:
A customer is having problems to boot these, see https://jel.ly.fish/ada08232-d2ce-4355-b41f-273e129fa146 

https://www.arrow.com/en/products/945-13541-0000-000/nvidia

Linked to the JF ticket so that we can come back to the customer once we have tested these devices.

acostach, 2020-11-17:
Currently shipping only from US.

@mpous will receive a unit, we'll test with that one once it arrives.

mckelvaney, 2020-11-17:
I'm in possession of several units in the UK. I'm happy to ship one to an engineer for testing if it will help speed things up.

geologic, 2020-12-04:
I've order a 2Gb nano, so if you need help for testing, please let me know

acostach, 2020-12-07:
@mpous did your nano 2GB arrive?

geologic, 2020-12-07:
> 
> 
> @mpous did your nano 2GB arrive?

Yes, if you need me to do tests, i'll be happy to help.

arronmabrey, 2021-01-08:
I just got mine today and have already found myself here, ha. Good luck y'all.

acostach, 2021-01-08:
@arronmabrey looks like this boot issue is not something that can be debugged remotely, I've already ordered one unit for myself to look into it but it didn't arrive yet. Will post updates here when it arrives and I get to have a look at it.

dtischler, 2021-01-22:
Mine just arrived yesterday, so I can test once ready.

dtischler, 2021-01-22:
Staging build booted right up, testing a container now.

acostach, 2021-01-25:
Hi @arronmabrey @geologic @mckelvaney an image for the 2GB Nano is now available for testing in [staging](https://dashboard.balena-staging.com) and will get in [production](dashboard.balena-cloud.com) too after we run the necessary test suites. Please note that the staging images are not validated and should not be used for production purposes.

mckelvaney, 2021-01-25:
Thanks for the update @acostach, will test it this week 👍

arronmabrey, 2021-01-25:
Yes thanks very much! I'll let you know as soon as I have something.

arronmabrey, 2021-01-25:
How do I access staging? My login/creds only seem to work for the production link/instance. I'm gonna try and reflash the image using the production link, but I'm guessing they haven't been updated yet and I'll get the same result.

dtischler, 2021-01-25:
Ah, very simple actually.  Staging does not share the same account / credentials as production, so, you will need to literally create a new account on staging.  Sign up with an email address and password, login, and go through the normal workflow of adding an Application and then a Device.  Mine did successfully run a workload, by the way.  :-)

arronmabrey, 2021-01-25:
@dtischler got it, I'm all signed in. The device booted and connected to the dashboard. 🎉  Thanks! What kinda timeframe does it take to get to production now?

mrpher, 2021-01-26:
Does the CLI need to be pointed to staging for a 'balena push' to work?  Currently have it hooked up to my prod account - cant imagine they are linked.

dtischler, 2021-01-26:
Correct @mrpher - you have to use something like this to split out Staging:  https://www.balena.io/docs/learn/more/masterclasses/advanced-cli/#13-separate-environment-configurations

NateDKing, 2021-01-27:
I have downloaded the image for the Jetson Nano 2GB off of the staging dashboard. With both the production and development editions, it is not showing up on the dashboard. I have plugged a display into it, and it stays on the Nvidia screen, and I have also tried pinging it, and it does respond. Any suggestions?

acostach, 2021-01-28:
@NateDKing you can attach a serial cable and try boot the development variant, see if it stops somewhere

acostach, 2021-01-29:
Device is now available in production at v2.67.3.

closed 2021-01-29 by acostach

NateDKing, 2021-01-29:
I connected a serial cable to the Jetson (945-13541-0000-000) with a fresh production / ethernet image from the non-developer website, and attached the log below. It took about 20 minutes, and seemed to keep exceeding retry counts. It did not show up in the dashboard. Part of the way down the log it stated "Net:   No ethernet found.", but then connected to the DHCP server. I tested with NVIDIA's image, and it was able to connect to with 100Mbps up / 10Mbps down.
Log: [putty.log](https://github.com/balena-os/balena-jetson/files/5896501/putty.log)

dtischler, 2021-02-03:
Ok Nate, this board is not booting from the SD Card, for some unknown reason.  You are not ever loading the balenaOS image that I assume you written to the SD Card...and instead the "fallback options" of USB boot and then Network boot are invoked. 

```
switch to partitions #0, OK
mmc1 is current device
Scanning mmc 1:e...
** Unable to read file / **
``` 

Are you certain you have written the image downloaded via balenaCloud dashboard to the SD Card properly?  Use Etcher to ensure it, and make sure validation is enabled to that it checks the SD Card afterwards.

Perhaps try another SD Card as well, just to see if that helps.  SD Cards can be a bit flaky in my personal experience, and the high quality (i.e., more expensive) ones are worth it.

NateDKing, 2021-02-04:
I have added 4 log filed below. I loaded the Linux operating system from Nvidia's website, to both a new 64GB Sandisk, and a new 64GB ADATA SD card. both times, the Jetson was able to find the operating system and boot. I then on the Sandisk card downloaded the production/ethernet version from the balenaCloud project, extracted it from the zip, and followed the same steps using balenaEtcher to write it to the card. And it showed the same error `** Unable to read file / **` when powered. I then proceeded with the ADATA card, and downloaded the developer/ethernet version this time, followed the same steps, and got the same results. All 4 times, balenaEtcher did verify the SD cards before I removed them.

[ADATA_Balena_Develeuper_Image.log](https://github.com/balena-os/balena-jetson/files/5927383/ADATA_Balena_Develeuper_Image.log)
[ADATA_Jetson_Linux_Image.log](https://github.com/balena-os/balena-jetson/files/5927384/ADATA_Jetson_Linux_Image.log)
[Sandisk_Balena_Production_Image.log](https://github.com/balena-os/balena-jetson/files/5927385/Sandisk_Balena_Production_Image.log)
[Sandisk_Jetson_Linux_Image.log](https://github.com/balena-os/balena-jetson/files/5927386/Sandisk_Jetson_Linux_Image.log)

acostach, 2021-02-04:
Hi @NateDKing , comparing the u-boot logs for Balena dev image and the Jetson Linux image, I see they both have:

`U-Boot 2020.04-g6b630d64fd `

If the u-boot was loaded from the Balena image, it should have shown:

`U-Boot 2020.10 (Nov 07 2020 - 11:31:11 +0000)`

If you take out the sd-card from the board and boot it, do you still see u-boot logs?

NateDKing, 2021-02-04:
Yes, when I boot after removing the SD card the log does still show `U-Boot 2020.04-g6b630d64fd`. 

[Jetson_Only Serial_Monitor_No Ethernet_No_SD.log](https://github.com/balena-os/balena-jetson/files/5928047/Jetson_Only.Serial_Monitor_No.Ethernet_No_SD.log)

acostach, 2021-02-05:
Hi @NateDKing , from what you describe, your Nano 2GB has u-boot written in the spi flash.

Maybe at some point you ran the NVida SD-card image for Jetpack 4.5 and during the initial setup you proceeded to have the QSPI to be updated? That tab mentions the newer qspi is not compatible with Jetpack 4.4. It's also possible that you nano came with the qspi pre-flashed for l4t 32.5.

I suggest you use the NVidia SDK manager to flash L4T 32.4.4 (jetpack 4.4.1) so that the compatible qspi image is written, and then try boot the Balena sd-card again.

NateDKing, 2021-02-10:
@acostach Thank you very much! I was able to load jetpack 4.4.1 onto the Jetson Nano, and it quickly connects to the dashboard after booting up.

acostach, 2021-02-10:
You're welcome, glad to hear it works fine now