balena-os/balena-jetson #141: [nano sd] spi issue reported on l4t 32.4.4

acostach, 2021-02-09:
spi loopback test from container is reported as failing with custom dtb on l4t 32.4.4, performed on the nano sd variant

awendel-presien, 2021-02-10:
Hi acostach,

Here is the dtb file we used to try and get SPI to work: [tegra210-p3448-0000-p3449-0000-b00-user-custom.zip](https://github.com/balena-os/balena-jetson/files/5955946/tegra210-p3448-0000-p3449-0000-b00-user-custom.zip). /dev/spidevx.x are present, but running a loopback test simply returns 0x00 instead of the bytes we send.

Thanks for your help.

Cheers,
Alex.

acostach, 2021-02-12:
Hi @awendel-presien , I suspect the dtb you attached may have been generated using an image based on an L4T different from 32.4.4. With the current dashboard image and the attached dtb the kernel hangs and reboots.

I generated a dtb with the 32.4.4 nvidia sd-card ubuntu image and jetson-io, and can see the issue. I also see this with the upstream Yocto demo image for 32.4.4. I've asked about it in upstream and will let you know as soon I receive updates.

awendel-presien, 2021-02-14:
Thanks @acostach , I'll double check regarding the dtb, but it's good you were able to reproduce the issue anyway. Let me know if you need any other information.

awendel-presien, 2021-02-15:
Thank you @acostach , I can confirm that works on our end, as well. Thank you for addressing this so quickly!

Is the solution you posted suitable as a permanent solution for 2.67.3? I noticed you'll be enabling SPI1 by default in 2.69.1.

acostach, 2021-02-17:
Hi @awendel-presien, we won't be releasing any jetson board on 2.67.3 anymore, instead we'll be using newer versions of meta-balena.

And now v2.69.1+rev1 for the jetson-nano sd-card is available in production, with spi1 enabled by default.

acostach, 2021-02-17:
Fixed in https://github.com/balena-os/balena-jetson/pull/144

closed 2021-02-17 by acostach

awendel-presien, 2021-02-18:
@acostach , that's fantastic. Thank you again for your help!