balena-os/balena-jetson #142: Health check incorrectly reports high temperature alarms

alexgg, 2021-02-10:
The health check on a customer's Nvidia Jetson Xavier NX Devkit SD-CARD with 8GB memory shows that thermal zone 4 (PMIC) is high in temperature. However, NVIDIA themselves (https://forums.developer.nvidia.com/t/thermal-zone4-reports-100-degree-celcius/44422/6?u=rakshak.talwar) say that the zone 4 temperature always reports as 100 degrees C, regardless of its actual value.

We need to confirm that this is the case, and if such modify the health checks not to report this as an error.

xginn8, 2021-02-16:
if this can/should be ignored, we can add any relevant slugs to the temperature check blacklist: https://github.com/balena-io/device-diagnostics/blob/master/scripts/checks.sh#L303-L317

acostach, 2021-05-17:
Fix is merged in device-diagnostics and available in production.

closed 2021-05-17 by acostach