balena-os/balena-jetson #166: CONFIG_BOOTPARAM_HUNG_TASK_PANIC causes panics with large images

jakogut, 2021-05-17:
The kernel is configured to panic on hung tasks, and detects balena engine as hung when syncing while writing a large image to disk.

```
[  102.104172] IPVS: Creating netns size=1920 id=6
[  223.808230] IPVS: Creating netns size=1920 id=7
[  243.565482] INFO: task balenad:4049 blocked for more than 120 seconds.
[  243.565737]       Not tainted 4.9.140-l4t-r32.4 #1
[  243.565853] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  243.566032] balenad         D    0  4049      1 0x00000008
[  243.566236] Call trace:
[  243.566354] [<ffffff80080863a4>] __switch_to+0x9c/0xc0
[  243.566479] [<ffffff8008f0d09c>] __schedule+0x22c/0x570
[  243.566590] [<ffffff8008f0d420>] schedule+0x40/0xa8
[  243.566744] [<ffffff8008f1057c>] rwsem_down_read_failed+0xd4/0x128
[  243.566872] [<ffffff8008f0f8a8>] down_read+0x58/0x60
[  243.566999] [<ffffff8008263568>] iterate_supers+0x78/0x138
[  243.567131] [<ffffff800829bac0>] sys_sync+0x50/0xc0
[  243.567237] [<ffffff8008083900>] el0_svc_naked+0x34/0x38
[  243.567394] Kernel panic - not syncing: hung_task: blocked tasks
[  243.567533] CPU: 3 PID: 47 Comm: khungtaskd Not tainted 4.9.140-l4t-r32.4 #1
[  243.567819] Hardware name: NVIDIA Jetson Xavier NX Developer Kit (DT)
[  243.568318] Call trace:
[  243.568508] [<ffffff800808c678>] dump_backtrace+0x0/0x1a8
[  243.571026] [<ffffff800808c844>] show_stack+0x24/0x30
[  243.576195] [<ffffff800841f4e8>] dump_stack+0x94/0xbc
[  243.581446] [<ffffff80081c3db0>] panic+0x128/0x28c
[  243.586599] [<ffffff8008178190>] watchdog+0x2f8/0x398
[  243.591760] [<ffffff80080dcd60>] kthread+0x100/0x108
[  243.596477] [<ffffff80080838a0>] ret_from_fork+0x10/0x30
[  243.602257] SMP: stopping secondary CPUs
[  243.605897] Kernel Offset: disabled
[  243.609511] Memory Limit: none
[  243.612484] trusty-log panic notifier - trusty version Built: 12:18:19 Oct 16 2020 [  243.636124] Rebooting in 1 seconds..
```

Increase the timeout, or disable `kernel.hung_task_panic` to remedy this.

thgreasi, 2021-05-28:
For reference (for other support agents), it seems that it depends on meta-balena v2.79.3.

Hades32, 2021-06-03:
Referencing https://jel.ly.fish/d46273c3-2d16-4a80-b3fb-52be338711a4

BTW I think we should not disable this. It only kills if a thread is `TASK_UNINTERRUPTIBLE` which I think is mostly due to hardware interaction. If this never recovers a reboot might be the only way out

jakogut, 2021-06-03:
@Hades32 This kernel config causes a panic and reboot upon detecting a task in uninterruptible sleep for longer than the configured timeout. 

The Kconfig help paragraph has this to say:
```
   Say Y here to enable the kernel to panic on "hung tasks",                                                             │  
   which are bugs that cause the kernel to leave a task stuck                                                            │  
   in uninterruptible "D" state.  

   The panic can be used in combination with panic_timeout,                                                              │  
   to cause the system to reboot automatically after a                                                                   │  
   hung task has been detected. This feature is useful for                                                               │  
   high-availability systems that have uptime guarantees and                                                             │  
   where a hung tasks must be resolved ASAP.                                                                             │  
                                                                                                                         │  
   Say N if unsure.  
```

In this instance, `balenad` is calling `sync(2)` and going into uninterruptible sleep waiting for the system call to return, and it doesn't before the timeout elapses because the write bandwidth of the storage medium can't handle the amount of data in the given timeframe. With this config enabled, the device reboots repeatedly and unexpectedly, without any sort of indication of why. As soon as the device boots back up, it starts pulling the images/deltas again, triggering the same behavior.

I think, if anything, this should be opt-in behavior, and it's still configurable using a sysctl tunable at runtime.

Hades32, 2021-06-03:
@jakogut why do you think we shouldn't increase the timeout and make that timeout configurable? The second support thread showed that removing this config didn't actually help. It now simply means that the device will be in the bugged state forever.

Also, how did you find out, that the storage medium is too slow? I try a 100MB dd on the device and it finished within a few seconds.

jakogut, 2021-06-03:
> why do you think we shouldn't increase the timeout and make that timeout configurable instead?

It's already configurable, with the `kernel.hung_task_timeout` sysctl tunable.

> The second support thread showed that removing this config didn't actually help.

I disagree. It makes the problem apparent in the kernel logs where otherwise we had a device that was difficult to debug as it was rebooting every couple minutes, and there was no indication as to why. After every reboot, there was no logged reason for the reboot, and no problem in the kernel logs.

Additionally, the customer reported that several of their affected devices _did_ succeed in deploying an application update after making the change, but not all of them.

> Also, how did you find out, that the storage medium is too slow? I try a 100MB dd on the device and it finished within a few seconds.

The affected user's images are ~8+ GB. The default hung task timeout is 120s, meaning the card would have to average greater than 68 MB/s sustained write to avoid triggering the hung task timeout. The Sandisk Extreme is  rated for ~70 MB/s, and the Extreme Pro is rated for ~90 MB/s, so it's close. However, it seems that something is stopping the kernel from returning from `sync` at all, and `cat /proc/meminfo | grep Dirty` shows:
`Dirty:               172 kB`

So it's possible there's a hardware problem or kernel bug going on here.

Regardless, defaulting to a reboot loop in this situation seems worse to me than writing a warning in the kernel log and leaving the device in a state conducive to debugging. If users want their devices to panic on detecting a hung task, they can still use the sysctl tunable at runtime.

Hades32, 2021-06-04:
Thanks for explaining @jakogut, I didn't realize we wouldn't have had the log otherwise.

The one thing I'm worried about: What if there are already devices which get hung sometimes and the reboot "fixes" the problem. Users probably wouldn't even notice, but after this change those devices would be in a bad state forever...

Can we at least add a grep for this kernel log to the device diagnostics, so we can detect that issue quickly?

jakogut, 2021-06-04:
> I didn't realize we wouldn't have had the log otherwise.

Sure thing. We're also interested in integrating the ability to persist kernel logs, including panics and oopses, in a persistent memory location through reboots. That feature would obviate the need to keep the system from rebooting here.

> The one thing I'm worried about: What if there are already devices which get hung sometimes and the reboot "fixes" the problem. Users probably wouldn't even notice, but after this change those devices would be in a bad state forever...

That's a good point, this has potential to leave some devices in a partially broken state, when a reboot would've otherwise patched it up enough for them to run. I'm not sure there's a perfect solution here, because rebooting unexpectedly degrades quality of service as well. I'd like to get to the root of the issue and figure out how to fix that, and we're working on it.

> Can we at least add a grep for this kernel log to the device diagnostics, so we can detect that issue quickly?

Dmesg output is already included in device diagnostics, and the hung task message is more of a warning than anything.

acostach, 2021-06-16:
Addressed in the NX SD image v2.80.3+rev1 (fix included in meta-balena). User is up-to-date of this.

closed 2021-06-16 by acostach