balena-os/balena-jetson #182: Sync syscall hangs indefinitely

jakogut, 2021-06-07:
A non-targeted `sync(2)` hangs indefinitely, even with few dirty pages and no in-flight write requests. 

To reproduce: run `sync` from the command line. It should exit with a zero return code shortly, but never does. A targeted `sync -f [path]` works as expected.

20k-ultra, 2021-06-07:
Board was running Host OS version balenaOS 2.67.3+rev5 production and full kernel version 4.9.140-l4t-r32.4

jakogut, 2021-09-28:
Tested for this with balenaOS versions:
```
2.45.1+rev3
2.56.0+rev4
2.69.1+rev4
2.73.1+rev4
2.82.11+rev8
```

But could not reproduce. If this issue comes up again, try to gather more information about the problem. Boot configuration application environment variables, etc.

closed 2021-09-28 by jakogut