balena-os/balena-jetson #207: Add CTI Photon TX2 NX device type

ClintHaerinck, 2021-08-10:
It would be great to have the CTI Photon TX2 NX as device type. I have the dtb's for JP4.5.1

acostach, 2021-08-12:
Sure! I think you can try to simply PR the dtb first, with a descriptive name like for instance cti_photon_tx2_nx_32_5_1.dtb etc, I'll include it and release the image and then you can set it in the device configuration table of a board that uses the default TX2 NX image and see if it works as expected. The TX2 NX uses u-boot just like the TX2 which, as opposed to the Xavier, allows for using custom dtbs without reflashing partitions, etc.

acostach, 2021-08-17:
Alternatively you can PR the device support as community by creating a coffee file, a yocto machine and the dtb.

ClintHaerinck, 2021-08-19:
I will make a PR now, is it possible to change the default dtb of the machine after the PR is merged? When not, please wait to merge the device type

acostach, 2021-08-20:
The default dtb can be changed by creating another PR after the original one is merged.

acostach, 2021-08-30:
Thank you for your PRs @ClintHaerinck, I'll close this ticket now that the device is available in production too.

closed 2021-08-30 by acostach