balena-os/balena-jetson #220: unable to build dev image

dremsol, 2021-09-20:
for some reason building development images using the flag in barys fails. It seems to have something to do with recent change in following [environment variable ](https://github.com/balena-os/meta-balena/commit/fc369799cdab0012cc9c75aaa4a2e54bb6b2656a). Altough following [commit ](https://github.com/balena-os/meta-balena/commit/cb1618b6d17a6c1118ea65d7485bb1bea907e18e) shows that development images can be configured at runtime. How do we enable development images before flashing? Seems to be done with correct setting in config.json in following [commit](https://github.com/balena-os/meta-balena/commit/5227ba18d0fe81fabae2b33b3f0cd9f690f85973)?

acostach, 2021-10-21:
@dremsol does the build still fail with v2.85.3? Automated builds work fine so if it still fails on your side please provide the steps used to reproduce the failure.

acostach, 2021-12-16:
Fix included in v2.88 release

closed 2021-12-16 by acostach