balena-os/balena-jetson #237: L4T R32.6.1

ClintHaerinck, 2021-12-09:
When is a move to L4T version R32.6.1 planned?

Can I still send PR's to add R32.5.X dtb's for current OS version?

acostach, 2021-12-09:
Hi @ClintHaerinck , currently it's work in progress, part of the work is already merged in the master branch on 32.6.1 so PRing 32.5.X dtb is no longer advised, unless they work for you with the 32.6.1 kernel. You can do a local build and check.

ClintHaerinck, 2021-12-13:
Is there a timeline when R32.6.1 should be available? Especially for the Jetson Xavier NX @acostach

acostach, 2021-12-15:
Which of them do you need first @ClintHaerinck ? The SD, eMMC or ReSpeaker?

ClintHaerinck, 2021-12-15:
It's for the Photon and CNX100 Xavier NX, which are all based on the eMMC version @acostach

acostach, 2021-12-15:
Ok, the Photon had it's dtb updated yesterday, the CNX100 will probably need an updated dtb from you for 32.6.1 and then I'll need to look whether the boot binaries need updating or some rework.

acostach, 2021-12-15:
and also for the Photon TX2 NX @ClintHaerinck

ClintHaerinck, 2021-12-16:
Is it possible to already provide an image for de eMMC devkit and Photon Xavier NX?

acostach, 2021-12-16:
@ClintHaerinck not at this time unfortunately, but you can try a local yocto build on your side

acostach, 2021-12-16:
also @ClintHaerinck please note that you'll  need to update the BSP link in jetson-flash too when writing your custom build. Let me know if something doesn't work with the master branch yocto build and I will try to have a look asap.

ClintHaerinck, 2021-12-16:
> @ClintHaerinck not at this time unfortunately, but you can try a local yocto build on your side

The build fails because it cannot find
```
photon-xavier-nx/bootlogo-photon-xavier-nx.blob
```

ClintHaerinck, 2021-12-16:
> > @ClintHaerinck not at this time unfortunately, but you can try a local yocto build on your side
> 
> The build fails because it cannot find
> 
> ```
> photon-xavier-nx/bootlogo-photon-xavier-nx.blob
> ```

I managed to build the image, had to clean the bootlogo-prebuilt package

acostach, 2022-02-07:
Hi @ClintHaerinck , I've triggered the deploys for the Xavier NX Devkit, Xavier NX Devkit eMMC and Photon NX, however, it's still necessary that you PR a 32.6.1 dtb for the [CNX 100](https://github.com/balena-os/balena-jetson/issues/237#issuecomment-994719669). Thank you

acostach, 2022-02-07:
Hi @KunalAivero, we'll also need a PR that updates the NX Respeaker dtb to L4T 32.6.1, so we can do a release for it too. Thank you

acostach, 2022-02-17:
Hi @ClintHaerinck @KunalAivero , any updates on the above?

KunalAivero, 2022-02-18:
Sorry for delay @ClintHaerinck will provide the new dtb soon

acostach, 2022-02-18:
Thanks and looking forward to the update PR.

Since support for all generic carrier boards is now available on L4T 32.6.1 in the master branch, all community device contributors are welcome to submit PRs in this repository to update their devices.

acostach, 2022-02-22:
Thanks @KunalAivero , I merged the update for the Seeed in #263 , will still need the CNX100 one.

acostach, 2022-04-13:
Closing as L4T 32.7.1 support has been merged in https://github.com/balena-os/balena-jetson/pull/284

@ClintHaerinck the CNX100 is missing a 32.6.1 release, so please update the dtb to 32.7.1 when the manufacturer releases it. Thank you

closed 2022-04-13 by acostach