balena-os/balena-jetson #247: photon-nano: SIM not recognised with modem in P3 (M.2 B-Key slot)

ClintHaerinck, 2021-12-23:
Maybe missing some kernel patches?

acostach, 2021-12-23:
we don't have or test the photon-nano hw but perhaps @pgils may be able to help here

pgils, 2021-12-23:
hi @ClintHaerinck can you provide some specifics of the modem? (make/model)
And is the modem itself recognized by the kernel?

ClintHaerinck, 2021-12-23:
The modem is recognized, and after a reboot sometimes not.
```
root@80b86fa:~# mmcli -m 0
  -----------------------------
  General  |         dbus path: /org/freedesktop/ModemManager1/Modem/0
           |         device id: ee1f98210dfa218fa71af5bbda793bf406a81459
  -----------------------------
  Hardware |      manufacturer: Sierra Wireless, Incorporated
           |             model: EM7565
           | firmware revision: SWI9X50C_01.09.04.00 6d3218 jenkins 2019/02/26 20:33:22
           |    carrier config: default
           |      h/w revision: 10001
           |         supported: gsm-umts, lte
           |           current: gsm-umts, lte
           |      equipment id: 359260080193987
  -----------------------------
  System   |            device: /sys/devices/70090000.xusb/usb1/1-2/1-2.2
           |           drivers: qcserial, qmi_wwan
           |            plugin: sierra
           |      primary port: cdc-wdm0
           |             ports: cdc-wdm0 (qmi), ttyUSB0 (qcdm), ttyUSB2 (at), wwan0 (net), 
           |                    ttyUSB1 (gps)
  -----------------------------
  Status   |             state: failed
           |     failed reason: sim-missing
           |       power state: on
           |    signal quality: 0% (cached)
  -----------------------------
  Modes    |         supported: allowed: 3g; preferred: none
           |                    allowed: 4g; preferred: none
           |                    allowed: 3g, 4g; preferred: 4g
           |                    allowed: 3g, 4g; preferred: 3g
           |           current: allowed: any; preferred: none
  -----------------------------
  Bands    |         supported: utran-1, utran-4, utran-6, utran-5, utran-8, utran-9, 
           |                    utran-2, eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, 
           |                    eutran-8, eutran-9, eutran-12, eutran-13, eutran-18, eutran-19, 
           |                    eutran-20, eutran-26, eutran-28, eutran-29, eutran-30, eutran-32, 
           |                    eutran-41, eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, 
           |                    utran-19
  -----------------------------
  IP       |         supported: ipv4, ipv6, ipv4v6
```

ClintHaerinck, 2021-12-29:
Tried other hardware and it seems the modem is broken.

closed 2021-12-29 by ClintHaerinck