balena-os/balena-jetson #285: L4T 32.7.1

acostach, 2022-04-13:
The L4T 32.7.1 PR https://github.com/balena-os/balena-jetson/pull/284 has been merged, so device-tree updates for community supported devices are welcome as soon as the dtbs are made available by the carrier board manufacturers cc @ClintHaerinck @KunalAivero @pgils @compiaffe @dremsol @abrayyan

compiaffe, 2022-04-13:
Thanks for the heads-up. Might take a while on my end.
Will this support Nanos?

ClintHaerinck, 2022-04-26:
@acostach Is it possible to make v2.95.15+rev5 of the CTI Photon TX2 NX available? I cannot find it, also not at balena-staging

acostach, 2022-04-27:
@ClintHaerinck 2.98.4+rev2 is now available in staging for the CTI Photon TX2 NX

ClintHaerinck, 2022-06-15:
Is it possible to make the photon-xavier-nx also available?

acostach, 2022-06-15:
Hi @ClintHaerinck, the community devices Photon Xavier NX and Photon Nano are now available at v2.98.33 - L4T 32.7.1

acostach, 2022-08-04:
Closing this ticket, L4T 32.7.2 support has now been merged in master. Updates for the community supported device-trees are welcome

closed 2022-08-04 by acostach