balena-os/balena-jetson #317: L4T 34.1 requested for CTI Rogue Xavier AGX

acostach, 2022-05-19:


acostach, 2022-08-09:
L4T 34.1.1 is a developer preview release which has some known limitations, like for instance no A/B rootfs support and should not be used in production environments. There will not be a production image available in the dashboard for this L4T version as L4T 35.x releases will be destined for production. cc @rosswesleyporter

SIGLOST, 2022-08-16:
@acostach nice progress! Where can I find the `L4T 34.1.1` image on DockerHub in order to start integrating it? Thanks

nmaas87, 2022-08-16:
@SIGLOST You can find it on Nvidia Website: https://developer.nvidia.com/embedded/jetpack-sdk-501dp - if I understand @acostach correctly, there will be no balenaOS release of 5.x until the first major / non-preview version of 5.x is available.

nmaas87, 2022-08-17:
Perfect, thanks so much for clarification @acostach - yeah I thought I read a lot changed along the sides of the other Ubuntu version and more stuff which makes it incompatible for a lot of entry Jetson boards. So thanks a lot for keeping up with the state of things! :)

dHofmeister, 2022-11-07:
Dear @acostach , 
I was wondering if we could have an update on the development of the Xavier NX r35.x base image? 

I see that the AGX Orin was released 2 weeks ago with r35.x support. Does that mean that the Xavier update is imminent? 

Greetings

SIGLOST, 2022-12-15:
Hello @acostach 
Is the  r35.x support on Jetson NX still on Balena's roadmap? Thank you

acostach, 2023-01-04:
Yes, the r35x is on our roadmap @SIGLOST but I don't have a timeline. Would you like to experiment with local yocto build until we finish the work on this?

volks73, 2023-01-24:
@acostach Hello! We would be interested in experimenting with the local yocto build as we are currently maintaining two separate fleets for our Jetsons: (1) Orins on r35 and (2) Xaviers (AGX and NX) on r32 and we would like to investigate merging these two fleets to single Jetson fleet.

acostach, 2023-02-03:
Hi @volks73, a few questions to better understand the context:

- For the AGX and NX ones, are you already using them already on L4T 32.7.x, provisioned as Jetson Xavier AGX / NX Devkit types in the cloud?
- Or will you be provisioning new Xavier AGX and NX devices only?
- In terms of actual hardware, are you using the Jetson Devkit carrier boards or some other types of custom / off-the-shelf carrier boards?

volks73, 2023-02-03:
Hello @acostach,

>For the AGX and NX ones, are you already using them already on L4T 32.7.x, provisioned as Jetson Xavier AGX / NX Devkit types in the cloud?

Yes, we have three (3) Jetson AGX Xavier dev kits and three (3) Jetson Xavier NX dev kits running L4T 32.7.x. These are spread between two fleets in balenaCloud.

>Or will you be provisioning new Xavier AGX and NX devices only?

We do not plan to provision any new Jetson AGX Xaviers or Jetson Xavier NX devices as we are focusing on the Jetson Orin line of devices.

>In terms of actual hardware, are you using the Jetson Devkit carrier boards or some other types of custom / off-the-shelf carrier boards?

In addition to our dev kits for the AGX and NX within the Jetson Xavier product line, we have 24 Jetson Xavier NX SoMs in a separate fleet across six [Jetson Mate carrier boards](https://www.seeedstudio.com/Jetson-Mate-Cooling-Kit-p-4784.html?queryID=dbbf20b91c3d9c04fe0b7703b89bfb85&objectID=4784&indexName=bazaar_retailer_products) with three to four SoMs per carrier board. Some of the Jetson Mates originally came with Jetson Nanos, but we have removed the Nano SoMs because they were under-powered for our application. All Jetson Xavier NX SoMs are the "eMMC" device type and running L4T 32.7.x across two fleets in balenaCloud.