balena-os/balena-jetson #389: Support for AGX Orin requested

acostach, 2022-08-04:


acostach, 2022-09-04:
Connects to https://github.com/OE4T/meta-tegra/issues/1055#issuecomment-1235920876

acostach, 2022-09-10:
Investigation temporarily paused, fix needed for https://github.com/OE4T/meta-tegra/issues/1055 - uefi_jetson built from sources crashes

edk2 nvidia from 34.1.1 cannot be used as a temporary solution because crashes with L4T 35.1

acostach, 2022-09-13:
@rosswesleyporter FYI ^

rosswesleyporter, 2022-09-13:
@acostach, thanks for the update

mikhail-vl, 2022-10-09:
@acostach https://github.com/OE4T/meta-tegra/issues/1055 was resolved. 

  Is anything else required to support Orin?

acostach, 2022-10-11:
Perfect, if that is solved we can now look into patching the new uefi bootloader for A/B rootfs selection and rollbacks on balena, generate the signed binaries for the Balena update bundle, etc.

I didn't get to test the upstream fix but I have an Orin AGX on my desk that runs a WIP local build of balenaOS with the uefi firmware built on the host, outside of Yocto. The uefi bootloader needs more patching for the above features so I can't promise a timeline but I'll post updates here once I'll have them.

acostach, 2022-10-12:
@mikhail-vl #415 is for Xavier Orin NX as the title says, this one is for the Xavier Orin AGX. Looking at https://www.nvidia.com/en-us/autonomous-machines/embedded-systems/jetson-orin/ I see these two are independent devices, but the Orin AGX can emulate the Orin NX. 

Currently I have the Orin AGX only and this is the one I'm currently working to add BalenaOS support for. Once this works fine with BalenaOS and is released to the cloud, I expect integration for the Orin NX to be faster since the two likely have many things in common, like bootloaders, partition layout, etc.

The bootloader issue I was referring to is the one that is marked as fixed in meta-tegra. https://forums.developer.nvidia.com/t/load-bootaa64-efi-from-a-different-fat-partition/229379 is different and I seem to have a workaround for it and that is not writing the kernel and dtb to their designated raw partitions. Still, I would like to look into that further and see what needs to be changed in the tegra UEFI firmware to boot the L4TLauncher directly.

acostach, 2022-10-18:
WIP PR in https://github.com/balena-os/balena-jetson-orin/pull/1

mikhail-vl, 2022-10-25:
@acostach Thank you for the update.

  1. We successfully connected Orin to our staging fleet.
  2. We already have Docker images for Orin, which we tested natively. I am pushing our first release as we speak.
  3. Display port does not work, and it's not required for our deployment.

  Well done!

mikhail-vl, 2022-10-25:
@acostach We also successfully built our own image from the `master` branch with the latest updates.

```
$ cat os-release 
ID="balena-os"
NAME="balenaOS"
VERSION="2.105.17"
VERSION_ID="2.105.17"
PRETTY_NAME="balenaOS 2.105.17"
DISTRO_CODENAME="kirkstone"
MACHINE="jetson-agx-orin-devkit"
META_BALENA_VERSION="2.105.17"
```

mikhail-vl, 2022-10-26:
Great job, team!

  The custom image based on the `master` branch https://github.com/balena-os/balena-jetson-orin is working as expected with NFS and GPU support.

  We connected it in development mode to the balenaCloud and will continue testing in the next two weeks.

acostach, 2022-11-18:
Hi @mikhail-vl, just wanted to let you know that the Orin is now available in the production cloud.

closed 2022-11-18 by acostach

mikhail-vl, 2022-11-18:
@acostach, well done! I am upgrading to the latest version and verify.

  Will let you know if we see any issues.

mmuhiedd, 2024-05-22:
Is there any update about the display port support? Trying to connect a screen using x11, xfce and nvidia-l4t-x11 without success so far ..

acostach, 2024-05-23:
@mmuhiedd does this [example](https://github.com/balena-io-examples/jetson-examples/tree/master/jetson-agx-orin-devkit) for xfce work on your AGX Orin Devkit?