balena-os/balena-jetson #415: Support for Orin NX requested

acostach, 2022-09-19:


acostach, 2022-09-19:
Linked pattern https://jel.ly.fish/pattern-customer-interest-nvidia-jetson-orin-nx-series-c5208c1

acostach, 2022-09-29:
BOOTAA64.efi isn't loaded by default on 35.1.0, question asked in upstream https://forums.developer.nvidia.com/t/load-bootaa64-efi-from-a-different-fat-partition/229379

acostach, 2022-11-01:
cc @mikhail-vl, I recall you were interested in testing this? ^

acostach, 2023-04-04:
Support for the Orin NX is being added in https://github.com/balena-os/balena-jetson-orin/pull/147

acostach, 2023-04-07:
The Orin NX 16GB module in Xavier NX Devkit (NVME boot) is now available in balena-cloud at v2.114.0

closed 2023-04-07 by acostach

mikhail-vl, 2023-04-07:
@acostach Thank you. We are using 32Gb and may explore 16Gb as they are available now in balenaCloud.