balena-os/balena-jetson #447: Not building after Ubuntu 22.04 update

ClintHaerinck, 2022-11-15:
Hi,

I'm getting following error when trying to build an image
```
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_image_hostapp_ext4
| 0+0 records in
| 0+0 records out
| 0 bytes copied, 1.0227e-05 s, 0.0 kB/s
| Loaded image ID: sha256:4c6b56054c42d3c07a4adb62f983b4faa0cd37343ba78293087841389f984c69
| + SYSROOT=/mnt/sysroot/inactive
| + pid=7
| + sleep 5
| + balenad -s=overlay2 --data-root=/mnt/sysroot/inactive/balena -H unix:///var/run/balena-host.sock
| balenad: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by balenad)
| balenad: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by balenad)
| balenad: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by balenad)
| sleep: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by sleep)
| Deleted: sha256:4c6b56054c42d3c07a4adb62f983b4faa0cd37343ba78293087841389f984c69
| Deleted: sha256:dc3d9c101f991294692c6f0603ca0e8e6031cbdcddb9311b2a169df3f62b5b37
| Deleted: sha256:62302780cc82da24f34656095d48ebe47c2df58a6e8e91b7bd2a275b2629a568
| WARNING: exit code 1 from a shell command.
NOTE: recipe balena-image-1.0-r0: task do_image_hostapp_ext4: Failed
ERROR: Task (/home/clint/git/balena/nano/balena-jetson/build-photon-nano/../layers/meta-balena/meta-balena-common/recipes-core/images/balena-image.bb:do_image_hostapp_ext4) failed with exit code '1'
```

Current installed version of GLIBC is 2.35. Am I missing some package or is something else wrong?

Kind regards,
Clint

acostach, 2022-11-21:
Hi Clint, the setup validated and used for building the cloud images is available at [balena-yocto-scripts](https://github.com/balena-os/balena-yocto-scripts/blob/master/automation/Dockerfile_yocto-build-env#L1)

It's possible that other setup variants may exhibit various errors or incompatibilities, therefore you should try using  the above setup on your host, or use that container as a base for building your local images.

acostach, 2023-01-03:
Closing due to inactivity. Feel free to re-open if the issue still occurs with the recommended setup.

closed 2023-01-03 by acostach