balena-os/balena-jetson #581: Microchip  mcp251x CAN driver not working on JetPack 4.6.3 and Nvidia Ubuntu

alexgg, 2023-04-05:
This has already been raised with Nvidia in https://forums.developer.nvidia.com/t/mcp2515-issue-thouroughly-followed-all-existing-threads-cant-get-it-to-work/246849.

In summary, balenaOS releases based on 32.7.3 (from v2.108.10) have a non-working CAN driver for this chipset.

acostach, 2023-11-30:
Hi @adolfogc , we don't support custom overlays for Jetson devices, only custom device-trees and this is applicable only for the devices which use u-boot in L4T 32.X. Please see https://docs.balena.io/learn/develop/hardware/i2c-and-spi/#jetson-devices for details on Jetson device-trees.

adolfogc, 2023-12-01:
Hi @acostach, thank you for your reply and for the information, I'll look into that documentation.